ivmech/ivport-v2 #1: self._bus.write_byte_data - IOError: [Errno 5] Input/output error

vahurtad, 2016-08-20:
Follwoing your guide, I get to this step:
```shell
 python init_ivport.py 
```
giving me back an error:
```
Traceback (most recent call last):
  File "init_ivport.py", line 6, in <module>
    iviic = IIC.IIC(addr=(0x70), bus_enable =(0x01))
  File "/home/pi/Desktop/ivport-v2/IIC.py", line 47, in __init__
    self._write(iic_register, config)
  File "/home/pi/Desktop/ivport-v2/IIC.py", line 50, in _write
    self._bus.write_byte_data(self._addr, register, data)
IOError: [Errno 5] Input/output error
```

Before I was getting a smbus error which I had to enable i2c in the raspi-config

How can I fix this problem?
My cables seem to be connected correctly.

ivmech, 2016-08-20:
Can you get address 70 when you send command

i2cdetect -y 1

-- 

Caner DURMUŞOĞLU
Mekatronik Mühendisi
*Ivmech Mekatronik İnovasyon Ltd. Şti.*
Cevizlidere Mah.1208. Cad. No: 4/5
Çankaya / ANKARA
0312 473 00 14
www.ivmech.com <http://www.ivmech.com/>

*Yasal Uyar (TR):*Bu e-posta yalnızca haberleşme amaçlı olarak sadece
ismi yazılı alıcının dikkatine yönelik olarak gönderilmiştir. İsmi
yazılı alıcı olmadığınız takdirde lütfen bu mesajı size gönderen kişiye
iade ediniz ve mesajın aslını imha ediniz. e-postada yer alan şirketi ve
alıcıyı ilgilendiren iş taahhütleri bağlayıcı değildir. Konfirmasyon;
şirketin resmi belgeler yoluyla antetli kağıda yazılmış yetkilinin
imzasının bulunduğu posta ya da faks yoluyla adrese yollanmış onayını
gerektirmektedir. Mesajın ekindeki tüm belgeler virüs kontrolünden
geçirilmiş olup virüslerden kaynaklanabilecek muhtemel zararlar sorumlu
olmayacağımızı ve alıcının belgeyi açmadan önce gerekli kontrolleri
yapmasını tavsiye ederiz.Teşekkür ederiz.
Bu mesajı yazdırmadan önce çevreye verebileceğiniz zararları bir kez
daha düşününüz.
*Disclaimer (EN): *This e-mail hereby was sent to the attention of the
intended recipient for communication purposes only. If you are not the
intended recipient please return this message to the sender and delete
the original message. The business commitments in the e-mail that relate
to the company and the recipient are not binding. The confirmation needs
the approval via official documents duly undersigned by the authorized
officer to be sent to the address by mail or fax on a stationary with
the letterhead. All the documents attached to the message were checked
for viruses; however the company shall not be responsible for any
probable damages due to viruses and we hereby recommend that the
recipient makes necessary controls before opening the document. Thank you.
Think of the environment once more before printing out this message.

vahurtad, 2016-08-20:
Thank you for the fast reply.
I did not get any address using i2cdetect.
Yes I am using ivport v2 and camera v2
I checked the cables + connecting the cameras using 
raspistill -o test.jpg
cables connected work, and so does the camera. 

I also get this error when trying to use the ivport and raspistill following these instructions found here: 
[ivport_test_A.py](https://github.com/ivmech/ivport/blob/master/ivport_test_A.py)

```

mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
mmal: Failed to create camera component
mmal: main: Failed to create camera component
mmal: Camera is not detected. Please check carefully the camera module is installed correctly
```

Am I connecting it wrong? Also my power is coming from a computer connected to the Raspberry Pi. Was reading that it was not enough power. I am away from it but I can try later.

lefsky, 2016-09-02:
I am getting the exact same set of messages with Camera V2 / and IVPort 2. Any resolution on this question? I am using a wall transformer

johnathanmarquez, 2016-10-08:
I get the same error at the same step:

`python init_ivport.py`

I'm using IVPort Rev 3 with two of the Pi Camera Rev 2 connected.  Red power LEDs on both cameras are on.
Please help, I have a time sensitive project dependent on this running.

johnathanmarquez, 2016-10-13:
I was originally misled by the label on the camera board I'm using.  While it reads "Raspberry Pi Camera (B) Rev 2.0", it should be clarified that it's actually using the 5MP OmniVision OV5647 sensor from the original official Raspberry Pi V1 camera module.

Here's a link:
[http://www.waveshare.com/rpi-camera-b.htm](url)

Could this be a reason for the issue?

ivmech, 2016-10-23:
We have added one more example with using ivport.py module for 4 camera supported ivport. You can check and test it, ivport type and jumper configurations should be defined while using ivport function from ivport.py module

https://github.com/ivmech/ivport-v2/blob/master/test_ivport_quad.py

Also please pay attention to that you are using "**ivport v2 multiplexer** with **raspberry pi camera module v2 (with 8MP sensor)**, if you are using 5MP camera module with ivport v2, there may be strange behaviors. 5MP camera module should be used with our old ivport multiplexer.

johnathanmarquez, 2016-10-31:
Hi, thanks for the response and the updates you've posted.

I've purchased two of the v2 camera modules and I'm running into the same issues.

When I get to the step `python init_ivport.py`
I get the same error [Errno 5] as in the beginning of this thread.

With i2c enabled in the raspberry pi settings, and checking with `i2cdetect -y 1`
I get nothing, all blanks.

However, with `vcgencmd get_camera`
I get supported=1 detected=1

I can't get past these steps.

Do you have any more suggestions?  Is it possible the ivport v2 module I received is defective?  Jumpers and pins are soldered correctly and the hardware is wired up correctly.

waqarrashid33, 2016-12-24:
I have the same problem, but it stops working after a few iterations. I have ivport v2 dual, raspberry pi 3, camera v2 8MP.
[image]

waqarrashid33, 2017-01-08:
I tried to solve this problem but all I couldn't find the reason for this error. Perhaps some hardware issue during switching. I did suppressed the error by making some changes in IIC.py, which is not a good practice but that is what I could think of. I also added an example to show video on screen instead saving pictures. I am attaching these filse here. I am also forking this repo and hopefully would do something with it.
[IOError.zip](https://github.com/ivmech/ivport-v2/files/692003/IOError.zip)

waqarrashid33, 2017-01-11:
I am not sure if this error has to do something with the Raspberry pi I2C clock stretching bug. But the randomness of this error suggest that it is.

http://www.advamation.com/knowhow/raspberrypi/rpi-i2c-bug.html
https://stackoverflow.com/questions/39274784/talking-smbus-between-raspberrypi-and-atmega-324pa-avr-not-clock-stretching
https://www.raspberrypi.org/forums/viewtopic.php?f=44&t=170958

vhurtad, 2017-02-27:
Hello everyone!
I gave up but I had someone help me through this a few weeks ago.
I fixed my issue by soldering the jumper "A" both in the front and back of the multiplexer. 

[image]
[image]

I had to do some digging through ivmech repositories to find out that you had to solder the jumper. Woops. Too bad they aren't helping anyone with any details!

Once you have cloned their repository, you need to upgrade and update your raspberry pi.
Next you need to enable I2C. I followed this tutorial :[Configuring I2C]( https://learn.adafruit.com/adafruits-raspberry-pi-lesson-4-gpio-setup/configuring-i2c)

Test your I2C and multiplexer after by using these commands :
[image]

init_ivport.py is to use the I2c address for the multiplexer.
Finally use the file test_ivport_quad.py to test the multiplexer that can use up to 4 cameras. (NOT THE DUAL which is used in  text_ivport.py)
Hopefully I can help someone fix their issues with this multiplexer.

eokeeffe, 2018-08-23:
I've been having problems with the dual board as well, i2detect -y 1 sometimes shows 

0 1 2 3 4 5 6 7 8 9 a b c d e f
00: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
10: 10 -- -- -- -- -- -- -- -- -- -- -- -- -- -- -
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: 70 -- -- -- -- -- -- -- -- -- -- -- -- -- --

It's odd, I haven't been able to get it working yet