ivmech/ivport-v2 #2: Cam. 2 Problem ~ IVPort Dual V2 Raspberry Pi Camera Module V2 Multiplexer

ThanasisPap, 2016-08-24:
I've already bought two of your Multiplexer modules (one for v1 camera and one for v2 -DUAL-). I read on your site that "There are some different videos while switching view source between 2 Different raspberry Pi Camera Module V2 with using IVPort Dual V2 module"  but I couldn't found them.

I tried successfully to run your test code from github, but I 've no image from the second camera. When it switches to camera 2 it returns a semi-destroyed frame but with image from the first camera. So I suppose that the module doesn't "see" the camera 2. 

To examine this I pull out the second RPi camera and the problem still remain. I got 2 pictures from the same first camera with the second destroyed ("snow" effect on image). 

**-I work on rpi 3 with V2 cameras.**

**-Above tests were implemented through your example code for image capturing -test_ivport.py-**
Is there some pin configuration that I've to change in order to see the right image?

Thanks a lot,
Thanasis Pap.

ivmech, 2016-08-24:
Hi Thanasis,
Was you module shipped to you with soldered header? if it is shipped with soldered header, it was fully tested before.
Aligning cable of camera module exists similar issues too, so please check its correct.

Also you can try `still_capture()` function which is shown below.

```
...
def main():
    #still_capture()
    #picam_capture()
    picam_sequence()
...
```
If problem continue, please let me know your order number and i can send you fully tested.
Best regards,
___
Caner

ThanasisPap, 2016-08-24:
Thanks for your fast answer!

I try still_capture() from the beginning. I install rasbian and make all the installation steps again. I have already enable I2C interface from raspi-config. The problem remains:

image 1: [https://drive.google.com/open?id=0B3WDu1S4C7AwNWg1MGhxTnlnY2s](img1)

image 2:  [https://drive.google.com/file/d/0B3WDu1S4C7AwNHlrYkJDbG94OXc/view?usp=sharing](img2)

The first camera (CAM1) sees the first image and the second was in dark in order to see the difference.
As you see I take picture from the same camera.

I did not change something in code, but only uncomment capture and comment sequence.
Is this code section need some change for DUAL V2?

`if camera == 1:
                if self.is_camera_v2: self.iviic.write_control_register((0x01))
                gp.output(self.fPin, False)
            elif camera == 2:
                if self.is_camera_v2: self.iviic.write_control_register((0x02))
                gp.output(self.fPin, True)`

ThanasisPap, 2016-08-25:
********************************************** SOLVED ***********************************************

The jumper 1 connection was not soldered as it supposed to be from delivery.
It works fine with RPi camera Module V2.

*******************************************************************************************************

closed 2016-08-25 by ThanasisPap