ivmech/ivport-v2 #3: Hardware/ setup issue with ivport-V2 board

Longans, 2016-09-29:
Please see anfegori91's post at https://github.com/ivmech/ivport/issues/21#issuecomment-250534439
It's got a nice picture of the issue. Re-posting here since our questions are about the V2.

Looks like my board has the R5-R6 shorted together. Are they electrically connected on that one side anyway?

I'm also confused with some of the jumpers, and the manual hasn't been much help. No idea what X and Y are, and if the GPIO (next to 3V3) needs to be jumper-ed as well.

I did try it out. the Pi can't see the camera with the mux unit in place. I have the two A jumpers shorted and have tried with and without the GPIO pins 7,11, and 12 set for each camera. Never got a connected camera.

Camera and all cables still work after taking the mux board out of the path.

Longans, 2016-09-29:
Update: Looks like that side of R5 & R6 both go to ground anyway, solder short wasn't the issue.

anfegori91, 2016-10-09:
Thanks, did you solved your camera connection problem? Have you used the A jumpers?

Longans, 2016-10-09:
No, I still haven't been able to connect to a camera. Both A jumpers are shorted the way the manual instructs.
Did yours work correctly?

ivmech, 2016-10-23:
We have added one more example with using **ivport.py** module for 4 camera supported ivport. You can check and test it, ivport type and jumper configurations should be defined while using **ivport function** from **ivport.py** module

https://github.com/ivmech/ivport-v2/blob/master/test_ivport_quad.py

anfegori91, 2016-11-15:
I've tested the board with four identical v2 cameras following the "getting started" guide and only CAM1 and CAM2 could be switched calling the `camera_change()` function. All connections were tested. Also, I've tried by changing the jumper to B and C with same results.  Note: when creating the iv object, the iv_jumper parameter was changed from `'A'` to`'B'` and `'C'` in my tests, but when getting the iv.f1pin and iv.f2pin values it still shows '11' and '12', so I had to change te code to force the pin change. @ivmech , why didn't you included `camera_change(3)` and `camera_change(4)` in the suggested example? is there a hardware issue in this board? (I've tried defining my own pin change functions but I can't get past the second camera!!!). @Longans how is your progress?

waqarrashid33, 2016-12-23:
Hey guys, I recently bought the IVPORT V2 Dual. I soldered the 13x2 pins to headers and then connected with raspberry pi. I tried to run the test_ivport.py but the results are weired.. It seems it is not switching. Previously I had V1.3 of these camears, due to these problems I bought two v2 cameras, but still no success. Is there any jumpers settings I had to make? Did you guys succeed in doing it? Would appreciate some help.

waqarrashid33, 2016-12-23:
Well I found out what I was doing wrong. I had to  enabled the Jumper 1. Shit now I don't have any soldering tools to solder it right now at home.

jdaniellhebert, 2017-01-04:
Following up on @anfegori91  comment:

I have experienced similar challenges and have yet to do the work to figure out how to get all 4 cameras working...it really has been a set back.  

If you guys (@ivmech) are not going to support your product it would be super helpful for you to give us schematics so we can figure out what is going on ourselves.

For example my goal is to have 6x cameras.  I am left guessing about how to connect these and my sense is that I am going to have to do a bunch of engineering to make it work...very different from expectations set on website.  It appears as if  you are not really monitoring this repository and replying in a timely manner.

So if you could update your examples to have 4 cameras work and then give instructions on how to do multiple boards that would be super helpful.  Schematics would be great too. Please.  Thanks for your consideration of this request.

jdaniellhebert, 2017-01-04:
Just to be clear the test_ivport_quad.py only works with 2x cameras (1 and 2).  The 3rd and 4th cameras do not work.  The type is QUAD2 (which seems to address the 2x camera on Quad board case)  and not QUAD (which would seem to address the 4x camera on Quad board case).  

In ivport.py for the Quad case:

else:
                print "Ivport type is QUAD."
                print "Cluster feature hasnt been implemented yet."
                print "There isnt camera: %d" % camera
                self.close()
                sys.exit(0)

jdaniellhebert, 2017-01-04:
Modifying your example1.py , I am able to run "raspistill -o still.jpg" in a separate shell.  It works for "c==1" and "c==2" but generates the following error for "c==3" and "c==4".

> mmal: No data received from sensor. Check all connections, including the Sunny one on the camera board

If you could please send a revised version of this sample that works, it would be very helpful.

Thank you

```
# adapted from https://github.com/ivmech/ivport/blob/master/ivport_test_A.py

import IIC
import RPi.GPIO as gp

iviic = IIC.IIC(addr=(0x70), bus_enable =(0x01))
iviic.write_control_register((0x01))                # default

gp.setwarnings(False)
gp.setmode(gp.BOARD)

gp.setup(7, gp.OUT)
gp.setup(11, gp.OUT)
gp.setup(12, gp.OUT)

gp.output(7, False)
gp.output(11, False)
gp.output(12, True)

c = ''

while c != 'q':
    c = raw_input("Enter Selection (q for quit):")
    if c == '1':
        iviic.write_control_register((0x01))
        gp.output(7, False)
        gp.output(11, False)
        gp.output(12, True)
    elif c == '2':
        iviic.write_control_register((0x02))
        gp.output(7, True)
        gp.output(11, False)
        gp.output(12, True)
    elif c == '3':
        iviic.write_control_register((0x04))
        gp.output(7, False)
        gp.output(11, True)
        gp.output(12, False)
    elif c == '4':
        iviic.write_control_register((0x08))
        gp.output(7, True)
        gp.output(11, True)
        gp.output(12, False)
    else:
        continue

gp.output(7, False)
gp.output(11, False)
gp.output(12, True)
```

waqarrashid33, 2017-01-07:
Hey, its nice to hear that you made it working.. I am using IVPORT DUAL V2 and I soldered jumper 1. While running the test_ivport.py for picam_sequence() I get this error:
self._bus.write_byte_data - IOError: [Errno 5] Input/output error
And I have no idea why is this error is arising. Perhaps due to trying to capture while the camera is not switched. Do you have any idea or did you faced this error?

jdaniellhebert, 2017-01-07:
@waqarrashid33 I have focussed on confirmation that the hardware is correct.  When I looked into the code, in particular the modified Picamera code, it became clear to me that there was too much complexity to debug be productive (it seemed incomplete and simple modification failed) and so focussed on modifying the simplest test program:  test_ivport_A.py.  When I did that I discovered a hardware bug (mine) that I was able to fix and then finished the test I show above.    

I also tried connecting multiple ivport boards, reverse engineered the hardware a bit and then concluded there is a hardware design error. I documented this in the Wiki page comments.  I have the sense that these board designs are not fully characterized and that testing is incomplete.

waqarrashid33, 2017-01-08:
I somehow managed to bypass the I/OError by adding try and except in the IIC.py. I think it is a hardware error something in the IIC mutiplexer as it doesn't able to switch in timely manner I guess.

jdaniellhebert, 2017-01-13:
I made some more progress and set up a repository with my work. 

https://github.com/jdaniellhebert/ivport_4.

I think i found the data sheet for the IIC mux. 

I would recommend you make a simple test software and verify the hardware.  My conclusion after all my efforts is that the IIC works well but the analog mux for the MIPI control signals is likely not fully characterized and there is likely variation across boards and configurations.  It's a pity we do not have schematics and layout.

anfegori91, 2017-02-15:
Hello, as I expected, my problem is a hardware one (had to see them with my usb microscope):
1. U2 pins shorted solder pads
[image]
2. U2 traces burnt.
[image]

Please IVMECH, I want a new board, I have tried by email, please answer.

jdaniellhebert, 2017-02-26:
@anfegori91 are you able to rework the boards?  I have pretty much given up on this hardware, it seems the IVMech have abandoned us.  I am currently working on another approach to mux cameras  for rpi and other cameras, perhaps if there is interest from others we could pool resources.

anfegori91, 2017-03-15:
I emailed this photos to IVMECH and they sent me another board. Tested and working well.

jackccwong, 2017-04-11:
root@ivport:~/ivport-v2 $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- 36 -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: 70 -- -- -- -- -- -- --

when python test_ivport.py with raspistill, it saved "still_CAM1.jpg" only and got error message
"mmal: No data received from sensor. Check all connections, including the Sunny one on the camera board"

Please advise!

waqarrashid33, 2017-05-05:
@anfegori91 Is it still working?
I checked mine for hardware faults but I couldn't find anything. I ignored the I/Oerror in software, but when I start receiving videos from both cameras, output from one camera gets full green or half green.
I researched it a lot and my conclusion was I2C clock streching bug in raspberry pi. Now I am using raspberry pi compute module. But just curious to know.

I didn't verified the I2C clock stretching error on serial analyzer. I will do it one day.

SloMusti, 2018-08-06:
@jdaniellhebert did you happen to succeed at the end with resin.io and this camera mux?

pranav1796, 2018-09-17:
> I've tested the board with four identical v2 cameras following the "getting started" guide and only CAM1 and CAM2 could be switched calling the `camera_change()` function. All connections were tested. Also, I've tried by changing the jumper to B and C with same results. Note: when creating the iv object, the iv_jumper parameter was changed from `'A'` to`'B'` and `'C'` in my tests, but when getting the iv.f1pin and iv.f2pin values it still shows '11' and '12', so I had to change te code to force the pin change. @ivmech , why didn't you included `camera_change(3)` and `camera_change(4)` in the suggested example? is there a hardware issue in this board? (I've tried defining my own pin change functions but I can't get past the second camera!!!). @Longans how is your progress?
> 
> log after failed attempt to change to camera 3 (a full tested brand new working camera):
> 
> ```
> mmal: No data received from sensor. Check all connections, including the Sunny one on the camera board
> mmal: mmal_vc_component_enable: failed to enable component: ENOSPC
> mmal: camera component couldn't be enabled
> mmal: main: Failed to create camera component
> mmal: Failed to run camera app. Please check for firmware updates
> ```

@anfegori91  I am facing a similar issue. I am able to get two cameras function but can't access cam3 and cam4. Is there any solution for this.

jdaniellhebert, 2018-09-18:
Hi sorry for the delay, I just saw this.  No I did not succeed and have
moved to another approach. My sense is this hardware has not been
adequately characterized nor supported...I imagine that this was a side
project and then there was not enough interest for them to invest more time
to take it further...

SteveNguyen, 2018-10-17:
Hi, I was trying to stack 2 ivport2 to get 5 cameras working but stumbled upon this thread.
I can get 4 cameras with one board following @jdaniellhebert but for now the stacking is still a mystery...
Is the conclusion is that we cannot stack ivport2 to get more than 4 cameras?