ivmech/ivport-v2 #4: backward compability?

WERAQS, 2016-10-01:
I'm out of my sony sensors and wanted to test it with old OV5647s' now getting this error, weird!

python test_ivport.py

Traceback (most recent call last):
  File "_ctypes/callbacks.c", line 314, in 'calling callback function'
  File "/home/pi/Desktop/ivport-v2/picamera/camera.py", line 129, in _control_callback
    "Received unexpected camera control callback event, 0x%08x" % buf[0].cmd)
picamera.exc.PiCameraRuntimeError: Received unexpected camera control callback event, 0x4f525245
Traceback (most recent call last):
  File "test_ivport.py", line 60, in <module>
    main()
  File "test_ivport.py", line 56, in main
    picam_capture()
  File "test_ivport.py", line 33, in picam_capture
    iv.camera_capture("picam", use_video_port=False)
  File "/home/pi/Desktop/ivport-v2/ivport.py", line 142, in camera_capture
    self.picam.capture(filename + "_CAM" + str(self.camera) + '.jpg', **options)
  File "/home/pi/Desktop/ivport-v2/picamera/camera.py", line 1619, in capture
    'Timed out waiting for capture to end')
picamera.exc.PiCameraRuntimeError: Timed out waiting for capture to end

etc. also sometimes something saying like "out of resources, not only memory"

jdaniellhebert, 2017-01-13:
I ran into this error initially.  My conclusion is that the problem is in this test code.  I reverted back to a much simpler example to test the hardware with some success.  They seem to say that the new board does not work with older camera though.  

I reverse engineered the boards a bit as well.

https://github.com/jdaniellhebert/ivport_4.

waqarrashid33, 2017-01-13:
I think the problem is with raspberry pi I2C clock stretching bug.

jdaniellhebert, 2017-01-13:
Are you experiencing this intermittently? I have seen similar yet less verbose error when hardware addressing is not set up correctly and multiple MIPI lines are connected to each other.  When I found the error it started to work repeatedly...

waqarrashid33, 2017-01-13:
I had not faced this issue. The issue I am facing is the I/OError which I tried to solve by modifying the IIC.py file but it is still not working properly as the images are not stable and goes back or green or something like that. The errors happen randomly which is kind of behavior the i2c stretching bug would cause. I hadn't tried reducing the i2c speed to check if it reduces the frequency of error as it is suggested for this bug.