ivmech/ivport-v2 #7: picamera.exc.PiCameraRuntimeError: Received unexpected camera control callback event, 0x4f525245

tuttoweb, 2017-05-04:
Dear all,

I have bought a IVPORT v2 multiplexer and I am having problem to make it work. Specifically, it worked once, then I rebooted and, after that, it did not work anymore. I get this error

$ python test_ivport.py 
Traceback (most recent call last):
  File "_ctypes/callbacks.c", line 314, in 'calling callback function'
  File "/home/pi/Desktop/Phenotiki/env/ivport-v2/picamera/camera.py", line 129, in _control_callback
    "Received unexpected camera control callback event, 0x%08x" % buf[0].cmd)
picamera.exc.PiCameraRuntimeError: Received unexpected camera control callback event, 0x4f525245
Traceback (most recent call last):
  File "test_ivport.py", line 60, in <module>
    main()
  File "test_ivport.py", line 56, in main
    picam_capture()
  File "test_ivport.py", line 33, in picam_capture
    iv.camera_capture("picam", use_video_port=False)
  File "/home/pi/Desktop/Phenotiki/env/ivport-v2/ivport.py", line 142, in camera_capture
    self.picam.capture(filename + "_CAM" + str(self.camera) + '.jpg', **options)
  File "/home/pi/Desktop/Phenotiki/env/ivport-v2/picamera/camera.py", line 1619, in capture
    'Timed out waiting for capture to end')
picamera.exc.PiCameraRuntimeError: Timed out waiting for capture to end

This is what I get with i2cdetect
$ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: 10 -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- 64 -- -- -- -- -- -- -- -- -- -- -- 
70: 70 -- -- -- -- -- -- --                         

and this with vcgencmd
$ vcgencmd get_camera
supported=1 detected=1

Everything should work, but apparently something is off and  I cannot figure out what is wrong.

Below you'll find a pic of the system. I chose the setup A as jumper.

[image]

Thanks a lot for your help

lrichard52, 2017-06-13:
I'm getting the same issue as well - really wish IVmech would provide some information on this issue. I can't get the board to work properly and everything is setup right. I'm getting the same error you are.

boyuanxiao, 2018-01-07:
DId you also try the test_ivport_quad.py script?

SandBC, 2018-05-24:
Hi everyone - running into the sqme problem; has anyone fixed the issue?

ggriffon, 2018-08-16:
I have the same issue

nicoceledon, 2019-01-24:
> DId you also try the test_ivport_quad.py script?

I had the same issue with test_ivport.py, but with the test_ivport_quad.py worked ;)

nicoceledon, 2019-01-30:
I'm having the same issue in my script. I haven't dig into it yet, but the ugly patch that i'm using now is init ivport.IVPORT and the closing it at the begining of my script. I'm using capture_continuous and camera capture in my script.