ivmech/ivport-v2 #10: supported=1 detected=0

tambetm, 2017-11-17:
I have IVPORT DUAL V2 connected to two V2.1 cameras on RPi 3 model B. I can detect both ivport dual and camera module successfully with i2cdetect:
```
tamme@tamme-desktop:~/ivport-v2$ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: 10 -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- 64 -- -- -- -- -- -- -- -- -- -- -- 
70: 70 -- -- -- -- -- -- --                         
```
But when I test for cameras they are not detected:
```
tamme@tamme-desktop:~/ivport-v2$ vcgencmd get_camera
supported=1 detected=0
```
What could be causing this? Internet suggests to check the ribbon cables, but I've triple-checked all cables dozen times and pretty confident they should be ok. Also both cameras work when connected directly to RPi.

Could it be that I didn't bother to solder the ivport card to gpio connectors? I just wanted to test it first and tilted it in such a way that all pins should be connected. The setup looks like this:

[image]

tambetm, 2017-11-21:
I soldered the board properly, but it still gave the same error. Then I did a reboot and it worked! So far it seems that:
1. I need to boot up raspberry.
2. Run init_ivport.py.
3. Reboot the system.
4. No need to run init_ivport.py second time, `vcgencmd get_camera` just works.

Not sure if the soldering helped or I just forgot to reboot the last time.

closed 2017-11-21 by tambetm

JamesCordell, 2018-04-26:
I found this out the hard way. Have you managed to git it working with out the reboot? this is a bit of a pain and not mentioned in the manual!

nortont, 2019-07-20:
This worked for me - do a reboot after init_ivport
I was restarting thinking that was the same

leoshmu, 2021-02-04:
@tambetm I have the exact same issue - I'm using a v2 dual with v2.1 cameras. I have connected it correctly as far as I can tell and my cameras alone work perfectly when connected to the rasbperry pi 4. The addresses appear correctly but the detected = 0. Perhaps it's a problem due to incompatibility with the raspberry pi 4? I'm really hoping someone can shed light on why it's not working!

tambetm, 2021-02-05:
@leoshmu unfortunately I don't have any new information. AFAIK the kernel needed the device to be activated, it couldn't detect it before. So I needed to run `init_ivport.py` first and then restart, then the device was visible to the kernel and was detected.