ivmech/ivport-v2 #11: Cluster ivport-v2, x8 camera(V2) does not working

Craig-Lee, 2018-01-26:
Cluster ivport-v2, x8 camera(V2) does not working
x8 cameras, x2 ivport-v2
ivport-v2(first layer) : 'A' jumper bridged
                                  'X','Y' jumper no bridged.
ivport-v2(second layer) : 'B' jumper bridged
                                        'X' jumper bridged.

But, ivport-v2(first layer)  works  without ivport-v2(second layer)

help me

ArvindNand, 2018-05-30:
were you ever able to find a solution to this problem?

SandBC, 2018-07-18:
Hello

 Did you find a solution for stacking two boards? 

 I define the two boards as 
 iv=ivport.IVPort(ivport.TYPE_QUAD2, iv_jumper='A') 
 iv2=ivport.IVPort(ivport.TYPE_QUAD2, iv_jumper='B')

 Then I try

 iv.camera_open()
 iv.camera_change(1)
 iv.camera_capture()
 iv.camera_change(3) ## (because I have hooked the second board to the port of the camera 3 on board 1) 
 iv.close() ## if i dont close this one before opening the other one it bugs out .. ?
 iv2.camera_open()
 iv2.camera_capture() ## and it takes a picture with the camera from the first board! Presumably having to close the first board is not ideal but otherwise I dont know how to open the second one.. 

 I also tried just doing iv.camera_capture with the port of the third camera, where the second board is hooked, and ah image comes up a part of which is the picture I want but it is really messed up..

 Any help or advice?!

Craig-Lee, 2018-07-19:
Hi, ArvindNand, SandBC

I couldn't find a solution with ivport...

So I made my own PCB borad with **FSA642 MIPI multiplexer** and succeeded.

Craig-Lee, 2018-07-19:
https://github.com/Craig-Lee

SandBC, 2018-07-19:
ok thanks! the more i try the more I think you cant really stack two of these boards..

Any chance you re commercializing your board ? :)

Craig-Lee, 2018-07-19:
I tried to [Arducam](http://www.uctronics.com/multi-camera-adapter-bundle-kits-compatible-for-official-raspberry-pi-v2-1pc-multiple-adapter-and-4-modified-v2-camera-module.html), But didn't work.
So, I asked the company(Arducam) a question. But they said only four cameras are possible.

If you want, I can send you the specifications of our product.

SandBC, 2018-08-17:
Hi Craig Lee

 That would be interesting yes. I created my own PCB but it does not seem to recognize the camera..

Thank you

SandBC, 2018-08-21:
Hi Craig, any chance you can send me info?