ivmech/ivport-v2 #12: Supported= 1 Detected= 0

CitizenDan, 2018-02-02:
I am getting this no matter which slot I try it in. We can read it but it won’t activate.

JamesCordell, 2018-04-26:
have you tried the python init_ivport.py then soft reboot via the command reboot then run cgencmd get_camera again?

TONINIZ, 2018-10-25:
I also had the same problem and I solved it by connecting the raspberry to the module and the module to the cameras. After that, I already showed the result Supported = 1 Detected = 1 and the 2 cameras that connected it worked for me, they did the multiplexing.

leoshmu, 2021-02-04:
Hello all , I am having the same issue with a pi 4. The cameras on their own are both v2.1, they both connect directly to the PI without trouble. I get the addresses to pop up correctly but the suppored=1 detected=0 is present even when I do the soft reboot as suggested by others. @ivmech is there something about pi 4 that is causing this error?

ghostiek, 2021-12-21:
> I also had the same problem and I solved it by connecting the raspberry to the module and the module to the cameras. After that, I already showed the result Supported = 1 Detected = 1 and the 2 cameras that connected it worked for me, they did the multiplexing.

@TONINIZ It's been 3 years so idk if you'll actually see this but can you please elaborate on this? I always need to reboot my RPi to run the test script and thats a problem