ivmech/ivport-v2 #13: Initalisation sequence

JamesCordell, 2018-04-25:
HI

I am finding my ivport-v2 works well however, I have to perform the following in order to get it to work.I am using r pi B.

I plug in the usb on the Pi and boot up.
I run.
vcgencmd get_camera
which returns
selected=1 detected=0
until I run init_ivport.py 
and then reboot again.
then I get 
selected=1 detected=1

from this point it works. Do you know if I can restart a service instead of having to reboot? Or why else I can do to get round this.

JamesCordell, 2018-04-26:
this is already mentioned by another person. 
https://github.com/ivmech/ivport-v2/issues/10
I will close this one.

closed 2018-04-26 by JamesCordell

nortont, 2019-07-20:
This worked for me - reboot needed